will127534/LoRa-concentrator #1: bad voltage regulator

nestorayuso, 2016-08-29:
Hi, 

I see in your schematic you are using a XC6204 voltege regulatos for 1.8 and 3.3 outputs from a 5v source.
This regulator has only 150mA max output current. This is insufficient.

SX1301 need 550mA typ, 750mA max @ 1.8v and 5mA typ, 10mA max @ 3.3v.

Even if you put a bigger LDO with 1A or more, the LDO will get very very hot, it will dissipate (5-1.8)*0.55= 1.76W of energy in hot.
I recommend to use a (step-down, DCDC converter...) switching regulator not linear regulator.

will127534, 2016-08-30:
Hi nestorayuso,
Thanks for pointing out the problem.
XC6204 is actually just for general SOT-23 LDO package, I solder RT9193-18PB for 1.8v and SPX3819M5-L-3.3 for 3.3v.

But you are right, it is insufficient for SX1301, I'll change the 1.8v power rail design to step-down next version. 

Thanks for the comment.

hallard, 2016-08-31:
For testing may be you can use dual output DC/DC step down like that, 
http://www.ebay.com/itm/DC-DC-Buck-Converter-Step-Down-Module-Power-Supply-1-5v-1-8v-3-3v-2A-Dual-Output-/262539347063

Not sure of quality but may be testing worth it.

will127534, 2016-09-02:
Hi hallard,
thanks for the advice, maybe I'll look for that IC, looks very nice (2-in-1 chip)
I'm using Lab power supply for testing though.
[image]
well I guess that SX1301 is dead, 1.8V rail only output 80mA, nowhere near the 700mA

This is the Log of util_pkt_logger:

loragw_pkt_logger: INFO: found global configuration file global_conf.json, trying to parse it
loragw_pkt_logger: INFO: global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
loragw_pkt_logger: INFO: lorawan_public 1, clksrc 0
lgw_board_setconf:361: Note: board configuration; lorawan_public:1, clksrc:0
loragw_pkt_logger: INFO: radio 0 enabled (type SX1255), center frequency 433000000, RSSI offset -166.000000, tx enabled 1
lgw_rxrf_setconf:415: Note: rf_chain 0 configuration; en:1 freq:433000000 rssi_offset:-166.000000 radio_type:1 tx_enable:1
loragw_pkt_logger: INFO: radio 1 enabled (type SX1255), center frequency 437000000, RSSI offset -166.000000, tx enabled 0
lgw_rxrf_setconf:415: Note: rf_chain 1 configuration; en:1 freq:437000000 rssi_offset:-166.000000 radio_type:1 tx_enable:0
loragw_pkt_logger: INFO: LoRa multi-SF channel 0 enabled, radio 1 selected, IF -400000 Hz, 125 kHz bandwidth, SF 7 to 12
lgw_rxif_setconf:519: Note: LoRa 'multi' if_chain 0 configuration; en:1 freq:-400000 SF_mask:0x7e
loragw_pkt_logger: INFO: LoRa multi-SF channel 1 enabled, radio 1 selected, IF -200000 Hz, 125 kHz bandwidth, SF 7 to 12
lgw_rxif_setconf:519: Note: LoRa 'multi' if_chain 1 configuration; en:1 freq:-200000 SF_mask:0x7e
loragw_pkt_logger: INFO: LoRa multi-SF channel 2 enabled, radio 1 selected, IF 0 Hz, 125 kHz bandwidth, SF 7 to 12
lgw_rxif_setconf:519: Note: LoRa 'multi' if_chain 2 configuration; en:1 freq:0 SF_mask:0x7e
loragw_pkt_logger: INFO: LoRa multi-SF channel 3 enabled, radio 0 selected, IF -400000 Hz, 125 kHz bandwidth, SF 7 to 12
lgw_rxif_setconf:519: Note: LoRa 'multi' if_chain 3 configuration; en:1 freq:-400000 SF_mask:0x7e
loragw_pkt_logger: INFO: LoRa multi-SF channel 4 enabled, radio 0 selected, IF -200000 Hz, 125 kHz bandwidth, SF 7 to 12
lgw_rxif_setconf:519: Note: LoRa 'multi' if_chain 4 configuration; en:1 freq:-200000 SF_mask:0x7e
loragw_pkt_logger: INFO: LoRa multi-SF channel 5 enabled, radio 0 selected, IF 0 Hz, 125 kHz bandwidth, SF 7 to 12
lgw_rxif_setconf:519: Note: LoRa 'multi' if_chain 5 configuration; en:1 freq:0 SF_mask:0x7e
loragw_pkt_logger: INFO: LoRa multi-SF channel 6 enabled, radio 0 selected, IF 200000 Hz, 125 kHz bandwidth, SF 7 to 12
lgw_rxif_setconf:519: Note: LoRa 'multi' if_chain 6 configuration; en:1 freq:200000 SF_mask:0x7e
loragw_pkt_logger: INFO: LoRa multi-SF channel 7 enabled, radio 0 selected, IF 400000 Hz, 125 kHz bandwidth, SF 7 to 12
lgw_rxif_setconf:519: Note: LoRa 'multi' if_chain 7 configuration; en:1 freq:400000 SF_mask:0x7e
loragw_pkt_logger: INFO: LoRa standard channel enabled, radio 1 selected, IF -200000 Hz, 250000 Hz bandwidth, SF 7
lgw_rxif_setconf:493: Note: LoRa 'std' if_chain 8 configuration; en:1 freq:-200000 bw:2 dr:2
loragw_pkt_logger: INFO: FSK channel enabled, radio 1 selected, IF 300000 Hz, 125000 Hz bandwidth, 50000 bps datarate
lgw_rxif_setconf:549: Note: FSK if_chain 9 configuration; en:1 freq:300000 bw:3 dr:50000 (50000 real dr) sync:0xC194C1
loragw_pkt_logger: INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
loragw_pkt_logger: INFO: gateway MAC address is configured to AA555A9000000000
loragw_pkt_logger: INFO: found local configuration file local_conf.json, trying to parse it
loragw_pkt_logger: INFO: local_conf.json does not contain a JSON object named SX1301_conf
loragw_pkt_logger: INFO: local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
loragw_pkt_logger: INFO: gateway MAC address is configured to AA555A9000000101
lgw_connect:524: INFO: no FPGA detected or version not supported (v0)
Note: success connecting the concentrator
setup_sx125x:495: Note: SX125x #0 version register returned 0x07
setup_sx125x:500: Note: SX125x #0 clock output enabled
setup_sx125x:557: Note: SX125x #0 PLL start (attempt 1)
setup_sx125x:495: Note: SX125x #1 version register returned 0x1f
setup_sx125x:503: Note: SX125x #1 clock output disabled
setup_sx125x:557: Note: SX125x #1 PLL start (attempt 1)
ERROR: Failed to load fw 1
ERROR: Version of calibration firmware not expected, actual:0 expected:2
loragw_pkt_logger: ERROR: failed to start the concentrator

Seems that PLL is locked!

nestorayuso, 2016-09-22:
Any progress?

550mA is when SX1301 is full running (8 MultiSF + LoRa + FSK receivers enabled), until not enabled 80mA can be a good value.

About PLL lock, 
do you have the 133MHz oscilator running?
do you have the 32MHz oscillator running and routed to both sx1255 radios?
which one of the sx1255 radios provides clock back to the SX1301?
is sx1301 configured to receive clock source from that radio?

will127534, 2016-09-27:
133Mhz and 32Mhz are both running, I can see both osc are working, alsothe 32Mhz output from SX1255 to SX1301, seems OK
Not sure for the 133Mhz osc though, I'm using LVCMOS/LVTTL compatible osc.
32Mhz is for both SX1255, and I connected Radio A's 32Mhz out to SX1301.

yuqian08, 2016-12-10:
Hi Will, this is very interesting board, we are working on design air quality sensors project based on Lora, but we doing the node with Microchip RN2903/2483, and have some productions released, maybe we can have some corporation in future :) 
I have a question, where you got those SX1301 documents and chips? from Semtech? thank you.

will127534, 2017-07-17:
Hi yuqian08 ,
I just noticed that Semtech release the full datasheet for SX1301 and SX1308.

will127534, 2017-09-01:
I've change 1.8V power reg to step-down, problem seems to be solved.
So closing this issue, thanks nestorayuso for pointing out this issue.

closed 2017-09-01 by will127534