will127534/LoRa-concentrator #2: ERROR: Version of calibration firmware not expected

will127534, 2017-01-13:
Currently this board still stuck on this problem,
Register Test passed, and I can confirm that the firmware is correct.
Here is the test_loragw_reg result.
Beginning of test for loragw_reg.c
Start of register verification
+++MATCH+++ reg number 0 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 1 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 2 read: 103 (67) default: 103 (67)
+++MATCH+++ reg number 3 read: 0 (0) default: 0 (0)
###MISMATCH### reg number 4 read: 204 (cc) default: 0 (0)
+++MATCH+++ reg number 5 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 6 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 7 read: 0 (0) default: 0 (0)
###MISMATCH### reg number 8 read: 20 (14) default: 0 (0)
+++MATCH+++ reg number 9 read: 0 (0) default: 0 (0)
###MISMATCH### reg number 10 read: 138 (8a) default: 0 (0)
+++MATCH+++ reg number 11 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 12 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 13 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 14 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 15 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 16 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 17 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 18 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 19 read: 1 (1) default: 1 (1)
+++MATCH+++ reg number 20 read: 1 (1) default: 1 (1)
+++MATCH+++ reg number 21 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 22 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 23 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 24 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 25 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 26 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 27 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 28 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 29 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 30 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 31 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 32 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 33 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 34 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 35 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 36 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 37 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 38 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 39 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 40 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 41 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 42 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 43 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 44 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 45 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 46 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 47 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 48 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 49 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 50 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 51 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 52 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 53 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 54 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 55 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 56 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 57 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 58 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 59 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 60 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 61 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 62 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 63 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 64 read: 0 (0) default: 0 (0)
[... 276 lines cut ...]
lgw_rxrf_setconf:453: Note: rf_chain 1 configuration; en:1 freq:433000000 rssi_offset:0.000000 radio_type:1 tx_enable:0 tx_notch_freq:129000
lgw_rxif_setconf:572: Note: LoRa 'multi' if_chain 0 configuration; en:1 freq:-400000 SF_mask:0x7e
lgw_rxif_setconf:572: Note: LoRa 'multi' if_chain 1 configuration; en:1 freq:-200000 SF_mask:0x7e
lgw_rxif_setconf:572: Note: LoRa 'multi' if_chain 2 configuration; en:1 freq:0 SF_mask:0x7e
lgw_rxif_setconf:572: Note: LoRa 'multi' if_chain 3 configuration; en:1 freq:-400000 SF_mask:0x7e
lgw_rxif_setconf:572: Note: LoRa 'multi' if_chain 4 configuration; en:1 freq:-200000 SF_mask:0x7e
lgw_rxif_setconf:572: Note: LoRa 'multi' if_chain 5 configuration; en:1 freq:0 SF_mask:0x7e
lgw_rxif_setconf:572: Note: LoRa 'multi' if_chain 6 configuration; en:1 freq:200000 SF_mask:0x7e
lgw_rxif_setconf:572: Note: LoRa 'multi' if_chain 7 configuration; en:1 freq:400000 SF_mask:0x7e
lgw_rxif_setconf:546: Note: LoRa 'std' if_chain 8 configuration; en:1 freq:0 bw:2 dr:16
lgw_rxif_setconf:602: Note: FSK if_chain 9 configuration; en:1 freq:0 bw:2 dr:64000 (64000 real dr) sync:0xC194C1
lgw_connect:528: INFO: no FPGA detected or version not supported (v0)
Note: success connecting the concentrator
lgw_setup_sx125x:407: Note: SX125x #0 version register returned 0x11
lgw_setup_sx125x:415: Note: SX125x #0 clock output disabled
lgw_setup_sx125x:469: Note: SX125x #0 PLL start (attempt 1)
lgw_setup_sx125x:407: Note: SX125x #1 version register returned 0xff
lgw_setup_sx125x:412: Note: SX125x #1 clock output enabled
lgw_setup_sx125x:469: Note: SX125x #1 PLL start (attempt 1)
ERROR: Version of calibration firmware not expected, actual:176 expected:2
*** Impossible to start concentrator ***

Still not sure were the problem is, by looking at the SPI bus it seems the data is right,
it really sent out 176 when reading register.
[image]

9600, 2017-01-29:
I saw something like this with an iC880A concentrator board when I accidentally had the current limited on a bench PSU set too low. It sort of appeared to work, but I got bogus messages about firmware versions being wrong and the packet forwarder would not start.

will127534, 2017-09-01:
I think the problem might cause by the reset line, I didn't have a resistor to pull down.
I left it floating during first couple time of testing,  and it might blow up the whole chip.

Anyway, this problem is not happened on my version 2, so closing this issue.

closed 2017-09-01 by will127534