will127534/LoRa-concentrator #3: Usage

alexanderplanar, 2017-06-27:
Hi! I'm interesting this project, I need to create a simple gateway for LoRa.
How to use this board? I mean, this is project, as I understand, provides the physical layer, what I should use  to connect this layer to LoRa server? 
For example, I found the project group [LoRa-net on github](https://github.com/Lora-net), I installed and setup LoRa-Gateway-Bridge, LoRa-Network-Server and LoRa-Application-Server on VirtualBox.
How to put it together?

mikini, 2017-07-06:
As I understand it, this board carries similar functionality as the [IMST iC880a concentrator](http://webshop.imst.de/ic880a-spi-lorawan-concentrator-868mhz.html) often used in [DIY TTN gateways](https://github.com/ttn-zh/ic880a-gateway/wiki) together with a Raspberry Pi. See the setup with a Pi in issue #1.

This board however is completely open hardware, but at the moment the design only carries a [sx1255](http://www.semtech.com/wireless-rf/rf-transceivers/sx1255/) (for 915MHz regions). For me located in Europe I'd need it tweaked for a [sx1257](http://www.semtech.com/wireless-rf/rf-transceivers/sx1257/) (868MHz) to be useful. I haven't looked into the details, yet, but are very interested too.

Seems like it is still under bring-up, though.

will127534, 2017-09-01:
Bring-up is completed guys, using version 2 hardware.
You can changed to SX1257 simply replace SX1255, it is pin compatible, but needs to change RF filter according to datasheet.

closed 2017-09-01 by will127534