will127534/LoRa-concentrator #5: Different GPS chip

will127534, 2017-09-04:
I thought that LoRa gateway support normal GPS with NEMA output,
but it seems to support only uBlox module, because it is using uBlox's UBX protocol for time adjustment.
So it might needs to change to uBlox module(MAX-7 series maybe) in the future.

mikini, 2017-09-07:
Hmm, yeah they use the identifier "ubx7" in [forwarder code](https://github.com/Lora-net/packet_forwarder/blob/master/lora_pkt_fwd/src/lora_pkt_fwd.c#L1102) and [gps implementation](https://github.com/Lora-net/lora_gateway/blob/master/libloragw/src/loragw_gps.c#L258) in which an unsolicited packet called UBX-NAV-TIMEGPS is enabled which is [explicitly parsed](https://github.com/Lora-net/lora_gateway/blob/master/libloragw/src/loragw_gps.c#L441).

The use of ubx is not even nicely abstracted into the libloragw but is exposed all the way up to the forwarder which defines a gps_thread [calling the ubx parse function explicitly](https://github.com/Lora-net/packet_forwarder/blob/master/lora_pkt_fwd/src/lora_pkt_fwd.c#L2745), so the stack seems rather hardcoded to uBlox GNSS'es at the moment and not trivially interchangeable with another way to get the time. Can't seem to grasp why the standard NMEA RMC time indication is not enough, though.

A quick look at the datasheet of [an M8-series uBlox](http://www.csgshop.com/product.php?id_product=178) that I've been testing it seem like it also supports the same TIMEGPS packet. But whether the details of parameters and data is the same as the 7-series I haven't studied.

will127534, 2017-10-03:
This UBX cmd should works on ubx6, ubx7 and ubx8 series chip, so variant of the module like Neo-6/7/8 Max-6/7/8 etc. should all works.
I've choose Max-7 series for ver2.1 though, I've used it for my high altitude balloon before, and some modules left in the part box.
Just tested ver2.1 with Max-7 and it works with the libloragw.

closed 2017-10-03 by will127534