will127534/LoRa-concentrator #10: Unreliable during heavy load

will127534, 2017-11-26:
Setup: 
one RF98 Transmitter @ 432Mhz, transmitting 6-axis data (24byte+2byte header) per 100ms
one RPI-Gateway with full channel on

If I tested indoor, Gateway will output some error data, the full log is here
[Log](https://asciinema.org/a/72G7waVnfnynvr8EB0Rmi2gFS)
it seems that SX1301 will output some wrong data.

But I when I try it near the window(with some wind), the error goes away, so I believed that it is cause by thermal problem, maybe do some measurement later.

I've already added a heatsink on top of SX1301, but it seems not enough....

apparell, 2018-03-08:
Hello, first of all congrats for the project.
I'm an energy engineering working in an heat-sink company. Maybe I can help you.
if did you record high temp, Can you post some value?

If the problem if thermal, we can design together an enclosure w/ embed heat-sink and ventilation.
let me know.
regards,
apparell

Brat0x2, 2018-07-14:
@will127534 Hi!
I've got the the board version 3 (2018/03/28). I noticed that sx1308 is very hot (can't touche my finger)! Is it normal working conditions?
There is no any heatsink on sx1308, the board powered from Raspberry Pi over 40pin connector, uUSB power source 5V, 2A.

will127534, 2018-07-14:
@Brat0x2 
Yes it is, SX1308 consume about 0.6A, which means it takes about 1W
I did measure about 55~60 Degrees on the SX1308, barely touchable...

will127534, 2018-07-14:
@Brat0x2
Depend on where if works, I have an indoor one without any heat sink and it works just fine,
But for my outdoor one, I did added a fan to cool two gateway in a box under the sun.

mikini, 2018-07-30:
You need to ensure proper transfer of thermal energy from both top (heat sink) and bottom (solder to PCB plane with substantial copper area 3 exposed to air) of the chip. To operate within the 0-70 deg. C operating conditions of the datasheet Semtech specifies 100 cm2 copper below and heatsink on top.

Let me quote from the datasheet:

> 2 Electrical Characteristics
> 2.1 Absolute Maximum Ratings
> 
> Parameter | Symbol|Conditions|Value 
> --|--|--|--
> Junction temperature |T(J,ABSMAX)|  |-40 °C  to 125°C
> 
> 2.3 Operating  Conditions
> The circuit will operate full specs within the following operating conditions.
> 
> Parameter | Symbol|Conditions|Min |Typ| Max |Unit
> --|--|--|--|--|--|--
> Ambient operating temperature |T(A)| With chip  paddle soldered to  PCB ground plan with minimum 100 cm2 air exposed area and heat sink |0| |70| °C

Source: [SX1308 Datasheet, paragraph 2](https://www.semtech.com/uploads/documents/sx1308.pdf#%5B%7B%22num%22%3A19%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C70%2C732%2C0%5D)

And from the thermal application note:

> 3. Practical Implementation and Measurements
> The recommendation of a 100 cm² PCB size area from the SX1308 datasheet comes from the simulation of the package heating dissipation, not from real measurements and seems most of the time quite unrealistic to achieve.
> In a practical situation, we need to ensure that the SX1308 junction temperature is always less than 125 °C (maximum junction temperature) inside its housing when the PicoCell Gateway is running at the maximum ambient temperature of its application.
> We strongly recommend mounting a heat sink on both top and bottom faces of the SX1308 chip for safe operation in all conditions. The energy dissipated is quite high. Keep in mind that the SX1308 chip can remain 100% of the time at 100% of clock speed and activity. The PicoCell GW is not a low-power device that only ramps up the clock from time to time, while absolute maximum ratings are for short term only.
> We selected the Mini Aluminum Heat Sink for Raspberry Pi - 13 x 13 x 3 mm / 0.5" x 0.5" x 0.1" (product ID: Adafruit Industries LLC 3084) for our Semtech reference design
 
Source: [SX1308 Thermal Dissipation: Technical Analysis and Recommendations, paragraph 3](https://www.semtech.com/uploads/documents/SX1308_PicoCell_GW_Thermal_Dissipation_Technical_Analysis_V1.0.pdf#%5B%7B%22num%22%3A12%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C43%2C745%2C0%5D)

will127534, 2018-08-05:
@mikini  That is a really useful data!
It seems that two Aluminum Heat Sink for Raspberry Pi (top and bottom) is good enough for heat dissipation. I think RPi Gateway V3 is larger than their PicoCell gateway, so using the same setup will perform better.

From the measurement on "RPIZero" Gateway V3(not released yet), I measured about 50~55 degrees with a large heatsink and 60 degrees without, which is better than PicoCell GW without housing and with heat sink. Although I didn't run for 24 Hours, only for couple hours. 

Anyways, I'll do a experiment on RPI Gateway V3 later.