will127534/LoRa-concentrator #11: SX1301 VS SX1308?

adz1122, 2017-12-08:
I would like to ask, what is the difference SX1301 and SX1308?
think you!

will127534, 2017-12-08:
In terms of function: No, SX1301 is same as SX1308, pin-to-pin and software compatible
But, SX1301 has -142.5 dBm sensitivity, SX1301 only have -139 dBm.

adz1122, 2017-12-08:
thinks，“But, SX1301 has -142.5 dBm sensitivity, SX1301 only have -139 dBm.”SX1301 only have -139 dBm?

will127534, 2017-12-08:
Sorry typo
Should be SX1301 has -142.5 dBm sensitivity, SX1308 only have -139 dBm.

adz1122, 2017-12-08:
thinks you， Are you doing your own gateway? I am also very interested in this, add a WeChat?
My WeChat：17512020988 
I am from ShenZhen An electronic enthusiast

will127534, 2017-12-09:
Yes, I am building by own gateway.
Shenzhen is a perfect place to be electronic enthusiast, too bad that I don't have WeChat though...
如果可以講中文的話就直接打中文好了 :D

adz1122, 2017-12-09:
好的，以后多交流，你开发这个网关准备用在哪里呢，我很想用LORAWAN做些传感器用在农业上，不过我没玩过树莓派，只玩过arduino或者STM32了。

will127534, 2017-12-09:
最主要應該是用在探空儀(High Altitude Balloon)，其他的話應該就小範圍的感應器網路了，類似你說的農業用途，不過收的應該是氣體感應器與PM2.5的資料
STM32應該也能玩網關，Semtech有一個[picoGW](https://github.com/Lora-net/picoGW_mcu)
雖然picoGW還需要用電腦，但是STM32<->SX1301的HAL都在那邊，應該可以做一個獨立的網關
我有看到一些掏寶上的店家在做AT-cmd的Gateway，我猜應該是拿picoGW改的

closed 2018-05-26 by will127534

Serjaru, 2018-10-05:
> 
> 
> thinks you， Are you doing your own gateway? I am also very interested in this, add a WeChat?
> My WeChat：17512020988
> I am from ShenZhen An electronic enthusiast

Can find 17512020988

bkrajendra, 2019-11-18:
Hi all Im also doing my own gateway.

goodmasher, 2020-04-21:
寡人是做射频硬件的，对于SX1257/SX1255 与 SX1301构成的网关，如何进工厂生产的测试一直找不到方法。希望大家在软件上能整个生产测试的方法。否则很难量产。