will127534/LoRa-concentrator #13: BOM and 868 mhz

karthirockz, 2018-03-23:
Hi Congrats and thanks for making it open source.

1. Can you please share the BOM file (bill of materials)
 
2. Please update LoRa-concentrator/Version2/RPI/SCH.pdf some of the labels of components are overlapped

3. What is the procedure to change it to 868 mhz

apparell, 2018-04-04:
To change the operational frequency you have to adapt passive RF components and SAW filter from the radio frequency front end. Ther's lot of resources on the web about sizing RF matching network also.
Regards!

apparell, 2018-04-04:
When Will will (WWW :D) publish the V3, I can help you to make 868 version. enjoy and happy open project!

clodpi, 2018-04-04:
@will127534 thanks, waiting for your 868 version

karthirockz, 2018-04-04:
@will127534 thanks, waiting for your 868 version

davidfungf, 2018-05-14:
Hi @will127534 , waiting for your 868 version too. Thanks.

closed 2018-05-26 by will127534

gadLinux, 2018-06-19:
Me too. I suppose it's not as easy as change the parts you told above