will127534/LoRa-concentrator #16: Can this module do LBT ? Listen before talk ?

4mig4, 2018-06-04:
It seems that HAL from SEMTECH supports LBT when version is > 0.4.1
But can AS923 be supported by your board ? (Japan)

will127534, 2018-06-04:
No, it won't support LBT since it needs a FPGA + SX1278 to scan the channel for LBT.

Note that if anyone have the ref V1.5 design sch, I'm willing to add LBT support. 😄

4mig4, 2018-07-15:
Ok thanks and no I do not have the reference design v1.5

will127534, 2018-08-31:
Reopen this issue because it is on my todo list...

piratfm, 2019-03-20:
Schematic can be reversed from this photos: https://fccid.io/AU792U13A16859/Internal-Photos/Internal-Photos-3321124

will127534, 2019-03-20:
Almost, but you can't, because it is a multi layer board.
I'm hoping to reverse by reversing the FPGA bitstream, but I can't read verilog......

piratfm, 2019-03-21:
Yep, You are right, but on this one:
http://www.embit.eu/wp-content/uploads/fronte-retro1.png
The signal layers are top/bottom and inner - are powers. So this is good suggestion to start. Also There is correct FPGA (iCE40LP1K-CM49) to which firmware is avaliable.

4mig4, 2019-03-21:
Very to see to that you think it is a worthy thing to do.

piratfm, 2019-03-22:
Pin mapping of FPGA, correct me if I'm wrong:
```
#SPI interface to HOST
set_io SPI_MISO      A5
set_io SPI_CS        A6
set_io SPI_SCLK      A3 #Wire shared with SX1301_SCLK and EEPROM_SCLK
set_io SPI_MOSI      A4 #Wire shared with SX1301_MOSI and EEPROM_MOSI

#SPI slave 0:
set_io SX1301_GPIO1      D7
set_io SX1301_GPIO3      A2
set_io SX1301_GPIO4      B2
set_io SX1301_CSN        B3
set_io SX1301_MISO       B4
set_io SX1301_RADIO_RST  F4 #this is input, not used in FPGA fw

#SPI slave 1: FPGA itselve

#SPI slave 2:
set_io EEPROM_CSN    A7
set_io EEPROM_MISO   C4

#SPI slave 3:
set_io SX1272_MOSI   A1
set_io SX1272_NSS    E2
set_io SX1272_MISO   D2
set_io SX1272_SCK    D3
set_io SX1272_RESET  F2

#Stays in the middle between I2S interface of SX1301 and SX125X (radio A)
set_io A_SP_CLK_IN   D1
set_io A_QI_TX_IN    B1
set_io A_IQ_TX_IN    C1

set_io A_SP_CLK_OUT  F3
set_io A_IQ_TX_OUT   G3
set_io A_QI_TX_OUT   G4
```
There is some small microchip SPI eeprom, but it's not used in software.
FPGA can be replaced by more DIY-friendly with QFN48 package, like ICE5LP1K-SG48. Only pin map file must be changed (and recompile firmware with icestorm). Also SX1272 share reference clock with SX125X and receiving RF just after second SAW (have no idea about how impedance matching has been made there). Maybe I can try to it with my lattice ICE40UP5K dev board later... But first need to solder (finalize) my gateway.

will127534, 2019-03-23:
@piratfm 
That was awesome!
I'm recently doing other projects that uses ICE5LP1K-SG48 (a.k.a learning verilog), so I also have some chip and boards :D 
I'm glad that I left a bunch of resistor between the SPI lines, it makes it easier to modify.
[image]

jgriessen, 2019-08-16:
Hi,
I have customers that want LoRa gateways.  I wonder if you are available to make some V3 boards
to sell me on tindie and coach me for pay on setup of all the software to connect to lorawan thethingsnetwork?  Later, we can make improved boards, weather enclosures, etc.  Interested?  Are you trained with a BSEE degree, or hobby?  Thanks, John

PostScript:  I knew some verilog for chip design, but have forgotten most of it.  Listen before talk would be a fine feature to brag about!

symdeb, 2019-10-22:
Hope Will would (re)consider to build another batch of boards. After surveying what is low cost and available right now, Lora, in low cost form, is still not so accessible, not much  choice and this looks to be the best solution so far.  RAK(831) boards are not cheap either. Trying to build an IoT gateway (not using RPI but with STM32/NXP based embedded with SPI) and would like to add in Lora. What would be nice are small companion device development ("thing") boards to connect sensors and actuators for prototype solution  purposes.  I am in Taipei City as well.

jgriessen, 2019-11-15:
@symdeb  How is your project to build IoT gateway with STM32/NXP parts coming?  Do you use pcb-rnd to make boards?  Maybe we can collaborate to get a low cost gateway and sensor boards?  I've done work with TinyOS on sensor boards, but that OS stalled.  RIOT is the OS I am interested in now.