will127534/LoRa-concentrator #18: one antenna

EduardShaid, 2018-06-13:
What is the reason that you switched from four antennas in version 1, to one antenna in version 3?

will127534, 2018-06-13:
The first version is just want to make sure SX1301 & SX1255 is working, so to keeping everything simple, every RF path is straight from IC to SMA connector.

After I successfully boot up SX1301,  the 2nd is aimed for 8-ch receiver, so I combine the both RX and 
ditch one Tx of SX125Xs, so two RF left, one TX one RX.

After the 8-ch receiver project is finished, there is some people asking for LoRawan support.
So to simplify the setup, I moved to single antenna, which combine RX/TX with RF switch and it is the current version.

closed 2018-06-13 by will127534

EduardShaid, 2018-06-19:
Thanks for the answer, but there are still questions:

1) The project design does not include the names of TXCO. Are there any requirements for their characteristics, such as accuracy or stability?
2) What were the requirements for the motherboard material or did the conventional FR-4 take it for the board?

will127534, 2018-06-19:
1.  I'm stick with Microchip Reference Design's NT3225SA, but I think any TCXO will do the job.
2. 4-Layer FR4 for version 3, I need more layers to maintain solid ground plane with RF frontends.