will127534/LoRa-concentrator #19: TXCO

EduardShaid, 2018-06-19:
There are still questions:

1. The project design does not include the names of TXCO. Are there any requirements for their characteristics, such as accuracy or stability?
2. What were the requirements for the motherboard material or did the conventional FR-4 take it for the board?

Thanks.

bkrajendra, 2019-11-18:
i have also asked same question at https://github.com/will127534/LoRa-concentrator/issues/38#issuecomment-552165267
But still waiting for reply from @will127534

will127534, 2019-11-18:
1.The main 133Mhz Osc doen't have a strict requirement, but the 32Mhz do, since it is the clock for RF components, the reference design uses 2725T series.
2.FR-4 will do the work.

closed 2019-11-18 by will127534