will127534/LoRa-concentrator #22: "failed to start the concentrator"

Zytheran, 2018-07-08:
I managed to get the system running and ran it say 5 times with no errors and was about to work out why the connection was not showing on the Things network when I got the error message "failed to start the concentrator".
command run was 
./lora_pkt_fwd > logfile.txt

Is there a list of things to check? It seems odd that is was working one minute and not the next. [The only thing I did was shut it down to swap over the power supply for the Raspberry Pi because I was getting low voltage warnings, which in itself was a bit odd because the voltage was 5.1V and it was a 2A supply providing 300mA. Anyhow swapped PS. I wasn't sure this was a Raspberry Pi voltage error and not one from the LoPoWAN board?]
Is there a list of things to check with the LoPoWAN setup, everything was done as per the instructions.
Up to the end of the TTN network setup guide.
log file is as follows:
*** Beacon Packet Forwarder for Lora Gateway ***
Version: 4.0.1
*** Lora concentrator HAL library version info ***
Version: 5.0.1;
***
INFO: Little endian host
INFO: found global configuration file global_conf.json, parsing it
INFO: global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
INFO: lorawan_public 1, clksrc 1
INFO: no configuration for LBT
INFO: antenna_gain 0 dBi
INFO: Configuring TX LUT with 16 indexes
INFO: radio 0 enabled (type SX1257), center frequency 917200000, RSSI offset -166.000000, tx enabled 1, tx_notch_freq 0
INFO: radio 1 enabled (type SX1257), center frequency 917900000, RSSI offset -166.000000, tx enabled 0, tx_notch_freq 0
INFO: Lora multi-SF channel 0>  radio 0, IF -400000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 1>  radio 0, IF -200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 2>  radio 0, IF 0 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 3>  radio 0, IF 200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 4>  radio 1, IF -300000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 5>  radio 1, IF -100000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 6>  radio 1, IF 100000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 7>  radio 1, IF 300000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora std channel> radio 0, IF 300000 Hz, 500000 Hz bw, SF 8
INFO: FSK channel 8 disabled
INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: server hostname or IP address is configured to "router.au.thethings.network"
INFO: upstream port is configured to "1700"
INFO: downstream port is configured to "1700"
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
INFO: GPS serial port path is configured to "/dev/serial0"
INFO: found local configuration file local_conf.json, parsing it
INFO: redefined parameters will overwrite global parameters
INFO: local_conf.json does not contain a JSON object named SX1301_conf
INFO: local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to B827EBFFFEE3F5F6
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
INFO: [main] TTY port /dev/serial0 open for GPS synchronization
ERROR: [main] failed to start the concentrator
-----------------------------------------------
Unfortunately I don't have a log from when it used to run but I know it was quite a few more lines, the GPS went on and off and then it was doing a few push commands or something.
json files attached.

[global_conf.json.txt](https://github.com/will127534/LoRa-concentrator/files/2174222/global_conf.json.txt)
[local_conf.json.txt](https://github.com/will127534/LoRa-concentrator/files/2174223/local_conf.json.txt)

Zytheran, 2018-07-10:
Solution was to go back into the lora_gateway directory
cd ./lora_gateway/

and then start the gateway again.

Start gateway by:
./reset_lgw.sh start

and then get back onto TTN via
cd ./lora_pkt_fwd
./lora_pkt_fwd

...and all good!

As mentioned in the opening post, this issue didn't always occur after power cycle however in the headless setup when the RPi is up a pole on the roof I'll be setting up a script to start the Gateway each time it boots before the LoRa packet forwarder starts.
(BTW issue with the noshow on TTN was because you need to configure the Gateway with the "Legacy packet forwarder" option when setting up a new Gateway)

will127534, 2018-07-10:
Hi Zytheran,
Yes, you should restart the gateway every power cycle to ensure SX1308 is in the init state.
I'll update the guide for clarify, thanks 👍

closed 2018-07-10 by will127534