will127534/LoRa-concentrator #23: GPS coordinates

Brat0x2, 2018-07-17:
@will127534 Hi!
I use GPS on the board version 3 with external antenna.
I see log messages:

### [GPS] ###
# Valid time reference (age: 0 sec)
# no valid GPS coordinates available yet
##### END #####

There is no valid coordinates for a long time. Why I didn't see them at all?

will127534, 2018-07-17:
Hi @Brat0x2 
It seems GPS cannot fetch enough signal for solving the location (but still able to output time reference)
Can you confirm the GPS antenna can get a good reception?

Also you can use
`screen /dev/serial0 9600`  
to see the NEMA output from GPS, which have some info about the signal strength.

Brat0x2, 2018-07-17:
@will127534 
[image]

will127534, 2018-07-17:
Yeah... It seems gps only lock  1 sat, where did you put the antenna?

Brat0x2, 2018-07-17:
@will127534 Simple external antenna with 2 meters cable, on the window to the North.

will127534, 2018-07-17:
It is active or passive antenna? 
if active, 5V or 3.3V?
The GPS Antenna port will output 3.3V to active antenna, so if you were using 5v active antenna,
it may cause the problem.

Brat0x2, 2018-07-17:
@will127534 It is active GPS antenna, 2.7...5.0V

will127534, 2018-07-17:
hmm.... can you put the antenna outdoor?

Brat0x2, 2018-07-17:
There is no possibility now. I'll try as I can. Thanks for your help!

Brat0x2, 2018-08-30:
@heitorPB the problem has been solved. GPS depends on our local weather.

closed 2019-11-18 by will127534