will127534/LoRa-concentrator #30: Component values

SG-O, 2019-03-25:
Thanks for your great design!

I'm currently picking all the parts to build my own full size V3. I have a few questions about some of the components:

L7 has a value of 8.2uH in a 0402 package. Is this really correct?
C21 is EOL and no longer available. Can I choose a different part? What are the requirements for the component?

Could you provide me with the components you use for botch oscillators? I have a hard time picking a correct model.

Thanks!

will127534, 2019-03-25:
L7 should be 8.2nH, I'll fix that typo, thanks!

C21 is decouple cap for GPS antenna power, so it is fine to replace with other similar cap,
just pick NP0 and it will be fine, the voltage rating for 47pF will be more than enough but just make sure it is larger than 6.3V

NT3225SA for 32Mhz TCXO
2725T for 133Mhz XO

will127534, 2019-04-05:
Yeah.. I think that's a little bit overkill, maybe replacing it with 100ohm will be fine.

piratfm, 2019-04-05:
The problem - is that SX1301 don't see SX125X chips on SPI bus at all with that loarge resistors. I replaced them to 0R and gateway starts working as expected.