will127534/LoRa-concentrator #31: Replacement for the GPS receiver

SG-O, 2019-03-25:
Would it be possible to replace the Max-7Q with the pin compatible Quectel L70. It outputs a  100ms 1 PPS signal but only supports NMEA output. 
This would significantly reduce the price (5$ vs 25$)

will127534, 2019-03-25:
Refer to issue #5
I did used Quectel L70 and L85 before, but I realize that the software is using uBlox's UBX protocol.
Max-7Q doesn't cost that much though.... there are a lot of module factory uses uBlox chipset to make compatible "Max-7Q"

mikini, 2019-03-25:
When I looked into this 1.5 years ago the software exclusively supported the ublox specific UBX protocol, see #5 (and similar question in #12). Don't know if that is still the case, however.

closed 2019-03-25 by SG-O