will127534/LoRa-concentrator #32: File Error

Cynicroute, 2019-04-23:
Getting `line 8, column 16: This is not an EAGLE file.` error with Radio_RPIv15_915.sch

mikini, 2019-04-23:
Maybe it would be a good idea to specify exactly in what software, version and computing environment you are attempting to open which edition of the file?

The only Radio_RPIv15_915.sch I can find in the current master branch seems to be generated by Eagle 8.4.3:

```
$ find LoRa-concentrator/ -name Radio_RPIv15_915.sch
LoRa-concentrator/Version3/RPI/Radio_RPIv15_915.sch
$ grep -r eagle LoRa-concentrator/Version3/RPI/Radio_RPIv15_915.sch 
<!DOCTYPE eagle SYSTEM "eagle.dtd">
<eagle version="8.4.3">
</eagle>
$
```
I don't know about Eagle or its file formats but line 8, char 16 in that file doesn't seem to contain anything suspicious.
```
$ head -8 LoRa-concentrator/Version3/RPI/Radio_RPIv15_915.sch |tail -1
<setting verticaltext="up"/>

$
```

Trying to import the Radio_RPIv15_915.sch and brd into KiCad 5.1 (File->Import Project->EAGLE CAD...) on Ubuntu 16.04 succeeds without any apparent issues (apart from some layer warning in brd) so I suspect the file must be ok.

Cynicroute, 2019-04-24:
Accessing /Version3/RPI/Radio_RPIv15_915.sch on Windows 10 Education, Build 17134.706 Version 1803. Using EAGLE 9.1.3 Education.

Indeed, there's nothing of note at L8,C16. Curious. A Google search did find other similar incidents with the same error. What worked for me was to unzip the entire branch and manually replace the .brd and .sch files with the stand-alone versions in the directory.

will127534, 2019-04-24:
I just download the file using right click on the "raw button"
and opened using EAGLE 9.1.1 on windows 10 1803, doesn't seems to have any problem though.

Cynicroute, 2019-04-24:
Must be something with my EAGLE build... I'll patch and update.

closed 2019-11-18 by will127534