will127534/LoRa-concentrator #33: Concentrator failed to start error

kungfupizza, 2019-05-15:
Hi,
I am having this error as shown below from a long time. It got solved once when a good power supply was provided but having it again. I always do the reset as told.
[image]

navinyate, 2019-05-15:
What is amps which u supplying to gateway. It should be atleast 2A

navinyate, 2019-05-15:
And check if there are any loose wires.

himitu23, 2019-05-15:
I also have this problem. I have searched several solution on some QAs, however I couldn't solve.

navinyate, 2019-05-15:
what kind of problem.

will127534, 2019-05-15:
LoRa Gateway is pretty sensitive to the power supply (normally talking to peak power consumption)
So make sure the power supply is working Ok.

navinyate, 2019-05-15:
Which is gateway ur using

himitu23, 2019-05-15:
> LoRa Gateway is pretty sensitive to the power supply (normally talking to peak power consumption)
> So make sure the power supply is working Ok.

OK, Thanks, I will try to use 2.5A supply

closed 2019-06-03 by kungfupizza