will127534/LoRa-concentrator #34: loragw_pkt_logger: ERROR: failed to start the concentrator

himitu23, 2019-05-15:
I have an error similar to that; https://github.com/will127534/LoRa-concentrator/issues/33

I have Version 3 of LoRa Concentrator, this is so good Gateway for using LoRaWAN with TheThingsNetwork for PoC.
I examined this on RPi3(Latest Raspbian Desktop), however it doesn't run successfully.

We can found this wiki; https://github.com/will127534/LoRa-concentrator/wiki/Startup-Guide
I got below error at first step(2.5A supplied);

pi@ttn-gateway:~/lora_gateway/util_pkt_logger $ sudo ./util_pkt_logger 
loragw_pkt_logger: INFO: found global configuration file global_conf.json, trying to parse it
loragw_pkt_logger: INFO: global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
loragw_pkt_logger: INFO: lorawan_public 1, clksrc 1
loragw_pkt_logger: INFO: radio 0 enabled (type SX1257), center frequency 915000000, RSSI offset -166.000000, tx enabled 0, tx_notch_freq 0
loragw_pkt_logger: INFO: radio 1 enabled (type SX1257), center frequency 916500000, RSSI offset -166.000000, tx enabled 0, tx_notch_freq 0
loragw_pkt_logger: INFO: LoRa multi-SF channel 0 enabled, radio 1 selected, IF -400000 Hz, 125 kHz bandwidth, SF 7 to 12
loragw_pkt_logger: INFO: LoRa multi-SF channel 1 enabled, radio 1 selected, IF -200000 Hz, 125 kHz bandwidth, SF 7 to 12
loragw_pkt_logger: INFO: LoRa multi-SF channel 2 enabled, radio 1 selected, IF 0 Hz, 125 kHz bandwidth, SF 7 to 12
loragw_pkt_logger: INFO: LoRa multi-SF channel 3 enabled, radio 0 selected, IF -400000 Hz, 125 kHz bandwidth, SF 7 to 12
loragw_pkt_logger: INFO: LoRa multi-SF channel 4 enabled, radio 0 selected, IF -200000 Hz, 125 kHz bandwidth, SF 7 to 12
loragw_pkt_logger: INFO: LoRa multi-SF channel 5 enabled, radio 0 selected, IF 0 Hz, 125 kHz bandwidth, SF 7 to 12
loragw_pkt_logger: INFO: LoRa multi-SF channel 6 enabled, radio 0 selected, IF 200000 Hz, 125 kHz bandwidth, SF 7 to 12
loragw_pkt_logger: INFO: LoRa multi-SF channel 7 enabled, radio 0 selected, IF 400000 Hz, 125 kHz bandwidth, SF 7 to 12
loragw_pkt_logger: INFO: LoRa standard channel enabled, radio 1 selected, IF -200000 Hz, 250000 Hz bandwidth, SF 7
loragw_pkt_logger: INFO: FSK channel enabled, radio 1 selected, IF 300000 Hz, 125000 Hz bandwidth, 50000 bps datarate
loragw_pkt_logger: INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
loragw_pkt_logger: INFO: gateway MAC address is configured to AA555A0000000000
loragw_pkt_logger: INFO: found local configuration file local_conf.json, trying to parse it
loragw_pkt_logger: INFO: local_conf.json does not contain a JSON object named SX1301_conf
loragw_pkt_logger: INFO: local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
loragw_pkt_logger: INFO: gateway MAC address is configured to AA555A0000000101
loragw_pkt_logger: ERROR: failed to start the concentrator

Who do know or can solve this?

will127534, 2019-05-15:
Ok, then try open the debug output to see where is the problem
to open the debug config, modify the following file: lora_gateway/libloragw/library.cfg
change DEBUG_HAL= 0 to DEBUG_HAL= 1
then recompile both packet_forwarder and lora_gateway and try running the gateway again.

himitu23, 2019-05-15:
Thanks for replying.

I got below;

pi@ttn-gateway:~/lora_gateway/util_pkt_logger $ ./util_pkt_logger 
loragw_pkt_logger: INFO: found global configuration file global_conf.json, trying to parse it
loragw_pkt_logger: INFO: global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
loragw_pkt_logger: INFO: lorawan_public 1, clksrc 1
lgw_board_setconf:427: Note: board configuration; lorawan_public:1, clksrc:1
loragw_pkt_logger: INFO: radio 0 enabled (type SX1257), center frequency 915000000, RSSI offset -166.000000, tx enabled 0, tx_notch_freq 0
lgw_rxrf_setconf:488: Note: rf_chain 0 configuration; en:1 freq:915000000 rssi_offset:-166.000000 radio_type:2 tx_enable:0 tx_notch_freq:0
loragw_pkt_logger: INFO: radio 1 enabled (type SX1257), center frequency 916500000, RSSI offset -166.000000, tx enabled 0, tx_notch_freq 0
lgw_rxrf_setconf:488: Note: rf_chain 1 configuration; en:1 freq:916500000 rssi_offset:-166.000000 radio_type:2 tx_enable:0 tx_notch_freq:0
loragw_pkt_logger: INFO: LoRa multi-SF channel 0 enabled, radio 1 selected, IF -400000 Hz, 125 kHz bandwidth, SF 7 to 12
lgw_rxif_setconf:607: Note: LoRa 'multi' if_chain 0 configuration; en:1 freq:-400000 SF_mask:0x7e
loragw_pkt_logger: INFO: LoRa multi-SF channel 1 enabled, radio 1 selected, IF -200000 Hz, 125 kHz bandwidth, SF 7 to 12
lgw_rxif_setconf:607: Note: LoRa 'multi' if_chain 1 configuration; en:1 freq:-200000 SF_mask:0x7e
loragw_pkt_logger: INFO: LoRa multi-SF channel 2 enabled, radio 1 selected, IF 0 Hz, 125 kHz bandwidth, SF 7 to 12
lgw_rxif_setconf:607: Note: LoRa 'multi' if_chain 2 configuration; en:1 freq:0 SF_mask:0x7e
loragw_pkt_logger: INFO: LoRa multi-SF channel 3 enabled, radio 0 selected, IF -400000 Hz, 125 kHz bandwidth, SF 7 to 12
lgw_rxif_setconf:607: Note: LoRa 'multi' if_chain 3 configuration; en:1 freq:-400000 SF_mask:0x7e
loragw_pkt_logger: INFO: LoRa multi-SF channel 4 enabled, radio 0 selected, IF -200000 Hz, 125 kHz bandwidth, SF 7 to 12
lgw_rxif_setconf:607: Note: LoRa 'multi' if_chain 4 configuration; en:1 freq:-200000 SF_mask:0x7e
loragw_pkt_logger: INFO: LoRa multi-SF channel 5 enabled, radio 0 selected, IF 0 Hz, 125 kHz bandwidth, SF 7 to 12
lgw_rxif_setconf:607: Note: LoRa 'multi' if_chain 5 configuration; en:1 freq:0 SF_mask:0x7e
loragw_pkt_logger: INFO: LoRa multi-SF channel 6 enabled, radio 0 selected, IF 200000 Hz, 125 kHz bandwidth, SF 7 to 12
lgw_rxif_setconf:607: Note: LoRa 'multi' if_chain 6 configuration; en:1 freq:200000 SF_mask:0x7e
loragw_pkt_logger: INFO: LoRa multi-SF channel 7 enabled, radio 0 selected, IF 400000 Hz, 125 kHz bandwidth, SF 7 to 12
lgw_rxif_setconf:607: Note: LoRa 'multi' if_chain 7 configuration; en:1 freq:400000 SF_mask:0x7e
loragw_pkt_logger: INFO: LoRa standard channel enabled, radio 1 selected, IF -200000 Hz, 250000 Hz bandwidth, SF 7
lgw_rxif_setconf:581: Note: LoRa 'std' if_chain 8 configuration; en:1 freq:-200000 bw:2 dr:2
loragw_pkt_logger: INFO: FSK channel enabled, radio 1 selected, IF 300000 Hz, 125000 Hz bandwidth, 50000 bps datarate
lgw_rxif_setconf:637: Note: FSK if_chain 9 configuration; en:1 freq:300000 bw:3 dr:50000 (50000 real dr) sync:0xC194C1
loragw_pkt_logger: INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
loragw_pkt_logger: INFO: gateway MAC address is configured to AA555A0000000000
loragw_pkt_logger: INFO: found local configuration file local_conf.json, trying to parse it
loragw_pkt_logger: INFO: local_conf.json does not contain a JSON object named SX1301_conf
loragw_pkt_logger: INFO: local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
loragw_pkt_logger: INFO: gateway MAC address is configured to AA555A0000000101
ERROR: FAIL TO CONNECT BOARD
loragw_pkt_logger: ERROR: failed to start the concentrator

Pin assign is failure?

will127534, 2019-05-15:
please also open DEBUG_REG
I forgot DEBUG_REG is the actual underlying interaction between SPI and SX1301.

himitu23, 2019-05-15:
I also got that(DEBUG_REG = 1);

pi@ttn-gateway:~/lora_gateway/util_pkt_logger $ ./util_pkt_logger 
...
loragw_pkt_logger: INFO: local_conf.json does not contain a JSON object named SX1301_conf
loragw_pkt_logger: INFO: local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
loragw_pkt_logger: INFO: gateway MAC address is configured to AA555A0000000101
ERROR CONNECTING CONCENTRATOR
ERROR: FAIL TO CONNECT BOARD
loragw_pkt_logger: ERROR: failed to start the concentrator

will127534, 2019-05-15:
Did you run ./reset_lgw.sh start before starting the gw?

will127534, 2019-05-15:
It seems like connection issue though...

himitu23, 2019-05-15:
Yes, I run "./reset_lgw.sh start" properly.
Of course, enabled SPI.

pi@ttn-gateway:~/lora_gateway $ ./reset_lgw.sh  start
Accessing concentrator reset pin through GPIO25...
pi@ttn-gateway:~/lora_gateway $ cd util_pkt_logger/
pi@ttn-gateway:~/lora_gateway/util_pkt_logger $ ls
Makefile  global_conf.json  inc  local_conf.json  obj  readme.md  src  util_pkt_logger
pi@ttn-gateway:~/lora_gateway/util_pkt_logger $ ./util_pkt_logger

himitu23, 2019-05-15:
Here is GPIO mapping;

pi@ttn-gateway:~ $ gpio readall
 +-----+-----+---------+------+---+---Pi 3B--+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 | ALT0 | 1 |  3 || 4  |   |      | 5v      |     |     |
 |   3 |   9 |   SCL.1 | ALT0 | 1 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 1 |  7 || 8  | 1 | ALT5 | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 1 | ALT5 | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |   IN | 0 | 11 || 12 | 0 | IN   | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   IN | 0 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |   IN | 0 | 15 || 16 | 0 | IN   | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 0 | IN   | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI |   IN | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO |   IN | 0 | 21 || 22 | 0 | IN   | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK |   IN | 0 | 23 || 24 | 1 | IN   | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 1 | IN   | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |   IN | 0 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 |   IN | 1 | 31 || 32 | 0 | IN   | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 |   IN | 0 | 35 || 36 | 0 | IN   | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 |   IN | 0 | 37 || 38 | 0 | IN   | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 0 | IN   | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+---Pi 3B--+---+------+---------+-----+-----+

navinyate, 2019-05-15:
Have u checked the NSS pin, is it connected to the right pin!

navinyate, 2019-05-15:
Is your local_conf.json file correct, because I don't see Ur device
connection to any lora backed server and still working with
global_conf.jskn file.

will127534, 2019-05-16:
Can you double check the soldering between the board and the pins?

himitu23, 2019-05-16:
Sorry, I cannot check because I don't have a tester..

himitu23, 2019-05-22:
It run succesfully with remounting the HAT, thanks.

closed 2019-05-22 by himitu23