will127534/LoRa-concentrator #38: LoRa-concentrator RPI Zero v3 BOM on octopart

bkrajendra, 2019-11-09:
Im trying to figure out right parts for this design.
Im sharing this BOM from Octopart, if any one can suggest if its ok, it will be good to buy the item.
https://octopart.com/bom-tool/LadeLROL

will127534, 2019-11-10:
OSC2 should be TCXO and package size should be 3.2mm x 2.5mm, and OSC1 should be 2.0mm x 2.5mm.
I use SiTime's product for the hight speed 133Mhz 2025 OSC.
Also, recommend changing SX1301 to SX1308 to save some cost.

Haven't checked the passive components but seems ok.

bkrajendra, 2019-11-10:
@will127534  thanks a lot for quick response.
I did replaced SX1308, but could not find exact part numbers for both OCX1 and OCX2
I found [this ](https://www.sitime.com/datasheet/2025)for 133Mhz , but it looks like SOT-23 package.
could no find 32mhz tcxo in 3.2x2.5mm type.

bkrajendra, 2019-11-18:
I did manage to find these two close match from microchips reference design:
OCX1 - https://www.ndk.com/images/products/catalog/c_2725T_lan_e.pdf - SKU - 644-1331-1-ND

OCX2 - https://www.jauch.com/downloadfile/57ff72b1b7707_2612e42791cc621776f7/jt33v_v6_171027_online.pdf - SKU - 1908-1045-1-ND

I hope these two make sense.

bkrajendra, 2019-11-18:
I just updated my BoM on octoparts.

will127534, 2019-11-18:
Just to note that the OSC size of the reference design is bigger than the Osc on the RPI-Zero version.
There isn't much space on the RPI-Zero version for me to squeeze the original Osc...