will127534/LoRa-concentrator #39: How adapt it for Frequency Plan IN865 for India

bkrajendra, 2019-11-09:
Hi i want adapt and make changes to this board for India IN865
Will it be sufficient to change following filters:
U$8,U$4,U$9  and
U$6
if yes what would proper value/model number for it.

and  do we need to change any other component.?

will127534, 2019-11-18:
I'm not sure which SAW filter you need to use for IN865
But just make sure that the passband for the SAW filter is in the range of  IN865.
Other components can stay the same.

closed 2019-11-18 by will127534

bkrajendra, 2019-11-20:
I found this one. 
http://52ebad10ee97eea25d5e-d7d40819259e7d3022d9ad53e3694148.r84.cf3.rackcdn.com/UK_ACT_ACTDF8137-865-0-QCC8C_DS.pdf

it has center frequency of 865Mhz

Dharm94, 2021-04-18:
@bkrajendra Brother did you tried this for IN865??? is it working??

bkrajendra, 2021-04-18:
not yet. COVID happened. and they left the work on this project due to all these chip market issues.
But I am willing to start it again. It's really nice project.
Currently, I am using RakWireless RPI GW.

I am planning to convert it to KiCad.
I Will update you if I get any success with this.