will127534/LoRa-concentrator #40: LoRa concentrator failed to start

kungfupizza, 2019-11-14:
Hi,
Since the production stopped, I printed my own PCB without no changes in the given schematics. When I test it, I still get "main: LoRa concentrator failed to start".

Am I missing something!?

bkrajendra, 2019-11-18:
Hi @kungfupizza  I am also trying to assemble my own board for rpi zero.
Can you help me regarding few component selection.
Im not finding OCX1 and OCX2 part numbers.
It will be great if you can tell me which parts number your using  or links to purchase for these two oscillators.
I can also help you in testing and resolving the issue.
You can find my entire list of BoM at :
https://octopart.com/bom-tool/LadeLROL
suggest any change if you think it can be made better.

will127534, 2019-11-18:
You can enable the debug output from lora_gateway/libloragw/library.cfg
And see which component failed.