will127534/LoRa-concentrator #42: Did anyone made one recently

wujiang, 2021-05-22:
First and foremost, huge thanks to the author @will127534 for all the work!

I'm trying to get the BOM ready and send the design to jlcpcb. Wondering if anyone made this recently so that we can share some resources.

asprakash, 2021-08-21:
Hi @wujiang .
I am also interested on this board. But unable to invest on this project since the main developer did not confirmed whether this board (as per the design files released) is working or not. However, I can fix the issues on the existing board since I am experienced on various boards like RAKWireless, NiceRF, iC880A, Sandbox,.etc. I need a board (with fully soldered) for testing purpose and I can fix the issues & make it work. May others can also join with us, @abhinav-2307 @kungfupizza @bkrajendra and others. 

TIA

bkrajendra, 2021-08-22:
I am also in.
We can either fork this repo or create a new one and start working on it. I am good with KiCad, we can start design review first and then can order some from JLCPCB. 
let make a group and start working.
let me know what you think.?

will127534, 2021-08-22:
Just adding comment that it is working - The board file released is the same as the one I sold on Tindle.
Also note that I have another board https://github.com/will127534/LoRa-GW-FeatherWing based on Kicad already.

asprakash, 2021-08-23:
Hi @will127534 Thanks for your comment and confirmation from your side. I will also go through the featherwing project. 

@bkrajendra Thanks for your interest. Instead of starting from scratch, let us focusing on the existing issues and how to solve them since @will127534 already confirmed that this PCB should be working fine when you are using the same design & components. Rest, the issues may be relevant to linux stuff. I would like to get one sample board from anyone instead of fabricating the PCB. Bcoz sourcing the entire BOM is taking more time, components shortage/delay everywhere.  May I know if you have one for testing. 

TIA

bkrajendra, 2021-08-23:
Yes, ok for me to focus on existing issues.
I am also struggling with components.
I will try if my Chinese vendor can supply it.

will127534, 2021-08-23:
IMAO, I think you need to change the DC-DC and LDO parts anyway because of the silicon shortages.

bhaktatejas922, 2021-11-18:
@will127534 How much did the board cost you to make ?

kongeo-aueb, 2022-02-24:
> Hi @wujiang . I am also interested on this board. But unable to invest on this project since the main developer did not confirmed whether this board (as per the design files released) is working or [...]

I am interested in asking you for some modufication advice on a Rak board.
I cannot contact you first
Here is my email kon123geo@gmail.com
Thank you in advance.

samuk, 2022-03-28:
Is anyone interested in moving this forward still? Did any boards get made? 

My personal interest is using the Pi Zero version with this [Banana pi Zero](https://forum.banana-pi.org/t/lora-for-m2-zero/13202) for a fully open hardware device.

How complex would it be to adapt to newer LoRa chips such as SX1278?

naturalmechanics, 2022-08-21:
I am interested.