will127534/LoRa-concentrator #43: Hi will

gadLinux, 2021-11-16:
I was using your board V3 with my rpi for a looong time.
I just migrated to ttnv3 and using t-impulse watch to do some tests. The packet forwarder seems to register right in TTN but when the watch tries to register we send back a bad message to the watch. I suppose that's a gateway trouble, because the libraries are open, ttn is tested by thousands and the only piece in the middle is the gateway...

arduino-lmic sees the message but ev_t ev value enum is strange and changes in every registration...

See something like 1257164: Unknown event     

Do you think that packet forwarder or gateway library is still ok?

will127534, 2021-11-16:
Can you lay down the software/library steps so I can reproduce or ask you to provide more logs by changing some library debug info?

gadLinux, 2021-11-17:
Hi Will, 
Yes, I followed the guide:
 * https://github.com/will127534/LoRa-concentrator/wiki/Startup-Guide
The Gateway is connected to TTN v3 and it shows it's live!
   https://ttnmapper.org/heatmap/private/?gateway=fe000100ff000001&network=NS_TTS_V3://ttn@000013
So that's a good thing
[image]

I can see Join request on TTN and in the gateway

JSON down: {"txpk":{"imme":false,"tmst":463689596,"freq":868.1,"rfch":0,"powe":14,"modu":"LORA","datr":"SF8BW125","codr":"4/5","ipol":true,"size":33,"ncrc":true,"data":"IOwUXX8rfaim5TUqK6X0jPGtg6gF4Xz0Q/16C4DLDq/P"}}

But it fails on device... because the kind of message is not expected.

I think that on device the code is pretty much standard:

https://github.com/Xinyuan-LilyGO/LilyGo-T-Impulse/blob/master/Example/T-Impulse-S76G-LoRaWAN/src/loramac.cpp

gadLinux, 2021-11-17:
Some logs too:
##### 2021-11-17 10:30:57 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 0
# CRC_OK: 0.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
# RF packets forwarded: 0 (0 bytes)
# PUSH_DATA datagrams sent: 1 (113 bytes)
# PUSH_DATA acknowledged: 100.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 6 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 0 (0 bytes)
# RF packets sent to concentrator: 0 (0 bytes)
# TX errors: 0
# TX rejected (collision packet): 0.00% (req:1, rej:0)
# TX rejected (collision beacon): 0.00% (req:1, rej:0)
# TX rejected (too late): 0.00% (req:1, rej:0)
# TX rejected (too early): 100.00% (req:1, rej:1)
# BEACON queued: 0
# BEACON sent so far: 0
# BEACON rejected: 0
### [JIT] ###
# SX1301 time (PPS): 92277009
src/jitqueue.c:448:jit_print_queue(): INFO: [jit] queue is empty
### [GPS] ###
# Invalid time reference (age: 1637145057 sec)
# no valid GPS coordinates available yet
##### END #####

will127534, 2021-11-18:
> But it fails on device... because the kind of message is not expected.
Any error log on the device side?

gadLinux, 2021-11-18:
Yes. But it's supposed to be tested with another gateway. So I better bet for a gateway problem because, protocol, outdated server, etc... Nobody reported something similar?
I was looking at the forum and see something about pushing wrong mensajes because temperature in the lora chip, I don't know if this can affect.

I will try debug device a little bit. Before getting again to the gateway. Can you confirm if the gateway correctly works with TTN v3? Anyone tested.

Thank you anyway for your support Will

gadLinux, 2021-11-20:
I added logs on device... And it seems nothing arrives...

Will check why

will127534, 2021-11-20:
> I will try debug device a little bit. Before getting again to the gateway. Can you confirm if the gateway correctly works with TTN v3? Anyone tested.

Nope, I haven't update my gw for a while, and I didn't tested with TTN v3.

I kind of wondering if it is related to pulse per second settings, does TTN v3 require accurate time pulse for downlink Tx?

gadLinux, 2021-11-20:
Finally I found the problem. And it was caused by me.
I added a delay in the main loop to see debug messages. And it delayed all processing missing the join messages. Sorry for causing troubles. 

And thank you again for your support. 

I can confirm the gateway and software is working with TTN3

closed 2021-11-20 by gadLinux