will127534/LoRa-concentrator #46: Is this project dead?

Andynopol, 2023-08-15:
I was just wondering if this project is still on going or not.

will127534, 2023-08-16:
[Wiki](https://github.com/will127534/LoRa-concentrator/wiki)

As for the original question: The project is not dead, it's finished. It has been running on the roof for some time now, and I'm satisfied with the board so no further dev is needed.

closed 2023-08-16 by will127534

Andynopol, 2023-08-16:
Sorry. Is this concentrator able to receive signals from multiple bandwidths(125kHz, 250kHz, 500kHz) and all spreding factors (from 7 to 12) at the same time?

This is what I am assuming reading the CONCENTRATOR term, so I just want to be sure.

And also I would like to know where can I order one?

mikini, 2023-08-16:
At least it has been available from Will on Tindie in the incarnations below;

* [Raspberry Pi LoRa Gateway](https://www.tindie.com/products/will123321/sx1308-raspberry-pi-lora-gateway-board/)
* [Raspberry Pi LoRa Gateway, bare PCB](https://www.tindie.com/products/will123321/bare-pcb-for-rpi-lora-gateway-v3-two-boards/)
* [Raspberry Pi Zero LoRa Gateway (smaller PCB by using single chip amplifier)](https://www.tindie.com/products/will123321/sx1308-raspberry-pi-zero-lora-gateway-board/)

To my knowledge not currently sold anywhere, but I have previously seen copies/variations of them available from different manufacturers. I have not followed LoRa development closely, but I guess new chipsets could have been released making this design obsolete/uninteresting. Maybe Will can answer that?

The Tindie product page/reviews contain some valuable information, but see also these threads;
* https://www.thethingsnetwork.org/forum/t/finished-opensource-raspberry-pi-gateway-hat-sx1301-8-base/10994/1
* https://www.thethingsnetwork.org/forum/t/sx1308-pi-gateway/13047/2

The Raspberry Pi GPIO/J8 header is [electrically identical across "1 Models A+ and B+, 2 Model B, 3 Models A+, B and B+, 4, and Zero WH and Zero W 2"](https://en.wikipedia.org/wiki/Raspberry_Pi#J8_header_and_general_purpose_input-output_(GPIO)), although probably there could be potential mechanical fitting issues lurking.

Andynopol, 2023-08-16:
The SX1303 is still used in products like Rak5146. So the obsolete problem I believe is far from being a concern

will127534, 2023-08-17:
Yeah why not? Is it based on SX1301/SX1308.

will127534, 2023-08-17:
For newer chipset though it is for another project: [https://github.com/will127534/LoRa-GW-FeatherWing](https://github.com/will127534/LoRa-GW-FeatherWing)

I don't see the point using RPI for LoRa GW so I end up building one for more MCU orientated board, but you can use it on RPI (I did that for HW testing)

Andynopol, 2023-08-17:
I understand. For now I'll try this one and if I am not satisfied with the results, I'll just get the other one. Right now I am trying to order the board on [https://jlcpcb.com](JLCPCB) and I have Total 72 parts detected ,38 Parts confirmed, 3 parts Inventory shortage, 31 parts not selected. Do you have another recomandation from where can I order it or the BOM and CLS files needs an update?

will127534, 2023-08-18:
It looks like you are not that familiar with HW from scratch, so I highly recommend you look at off-the-shelf options like the Rak5146 you've mention. 

This project, in general is to serve as the base for custom gateway design, like if you want a different frequency for some weird reason, or different board formfactor for other SBC/MCU boards or just you want to integrate fully to some design.

I won't update the BOM and CPL because the batch I've sell on Tindle is build with that BOM and CPL, this serves as a reference for those who've brought the board and want to check the parts. Also the format is not for JLCPCB (Yeah I know, but this is before JLCPCB PCBA have the chips I need for this build) but other PCBA factories.

In general, you might just use Rak5146 or something from RAK that fits, it will be easier for you to use. This project is for those who likes/needs to tinker with the GW design.

Andynopol, 2023-08-19:
I have a RAK5146 board. The problem that I have with it is that I am trying to connect to a personal application. Not Chirptsack/TTN. The problem that I have with this product is that is overly complicated when it comes to connecting to someple personal projects(is just a gateway, not a damn company with departaments and services - that is a small part of my frustration with this board). The documentation is a royal mess. The RAK5146 has the least amount of documentation compare to the 2XXX and 7XXX lineup.

That is why I was aiming for an opensource version.
Plus, there are bonus points if I don't use an off the shelf version for my final product on my exam.

Note: I was not expecting you to change the BOM in the repo, but to check if you can send me an updated version for JLCPCB or whatever PCBA factory.

I will try on my own. I was just trying to cheat a little bit of time...I guess. Sorry!

will127534, 2023-08-20:
The thing is, you can run the libloragw and it's helper program on those HW (RAK5146 ) that you don't have to connect to The Things Network. Probably just need to check the GPIO connections on the SX1303 and you are good to go and use either USB or SPI.

You will need to first understand the structure of these LoRa GW (SW and HW) before you proceed this route. And there is no point of catching the bonus point if you can't even get the thing to work at the first place.